TuragaLab/DECODE #3: Version tag should be automatically imported by conda and setup.py.

Haydnspass, 2020-09-30:
As described above. Can be extracted from decode/__init__.py
emitterset.save() should also store version tag because this might be needed in future.

olbris, 2020-10-14:
Fly EM at Janelia uses versioneer for this (https://github.com/python-versioneer/python-versioneer). It inserts version numbers based on git tags. I was going to investigate it for this project.

Haydnspass, 2020-10-23:
@olbris Done with v0.9.3-beta.2 (https://github.com/TuragaLab/DECODE/commit/a30e553f16e8d6e6a86b79eb39296afb2118e28a).
Used bump2version for this, it's the other way around it (1) bumps the version and  (2) creates a git tag/ version which is then to be pushed by `git push --tags`. I then edit the tag and upload the wheel.

closed 2020-10-23 by Haydnspass