TuragaLab/DECODE #85: import warning

Haydnspass, 2021-03-12:
I can reproduce this and get the same warning.

Here the warning that appears 2X8 times in an epoch when I set num+workers to 8 (8 times at training, 8 times at testing):

```
:\Users\chris\.conda\envs\decode_cuda\lib\runpy.py:127: RuntimeWarning: 'decode.neuralfitter.train.live_engine' found in sys.modules after import of package 'decode.neuralfitter.train', but prior to execution of 'decode.neuralfitter.train.live_engine'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
```

_Originally posted by @cleterrier in https://github.com/TuragaLab/DECODE/discussions/83#discussioncomment-473202_

Haydnspass, 2021-03-14:
@ASpeiser can you do this? Or have information on this?

ASpeiser, 2021-03-14:
Removing `import decode.neuralfitter.train.live_engine ` from the decode/neuralfitter/train/__init__.py fixes it for me. Does this break anything?

Haydnspass, 2021-03-15:
"Merged"

closed 2021-03-15 by Haydnspass