TuragaLab/DECODE #100: Doc changes

Haydnspass, 2021-03-17:
Refactor docs:
- clear installation instructions from index.rst --> new markdown installation.md
- add banner / nice image to index.rst 
- add youtube to index
- keep google colab at index, (add a few words)

Haydnspass, 2021-03-18:
It could be nice to add the youtube video (embedded HTML) to the Readme.md in the repo base folder as well. You can have a look at the existing code in tutorial and literally copy it.

Haydnspass, 2021-03-18:
@phoess that'd also be an option for the github readme (one can click on a button to open more detailed instructions)

https://github.com/PyTorchLightning/lightning-bolts/blob/master/README.md

Haydnspass, 2021-03-19:
Can you:
- add installation.md to the top bar (you should find somewhere how I put the Tutorial there but I don't remember ...)
- remove all the listings in index.rst
- rm installation instructions from landing page entirely but keep colab
- embed youtube as proper HTML (like in the current master in Tutorials)

phoess, 2021-03-19:
> It could be nice to add the youtube video (embedded HTML) to the Readme.md in the repo base folder as well. You can have a look at the existing code in tutorial and literally copy it.

Unfortunately, Github doesn't allow embedding. For the readme, I added a screenshot including a link to the video.

phoess, 2021-03-19:
> @phoess that'd also be an option for the github readme (one can click on a button to open more detailed instructions)
> 
> https://github.com/PyTorchLightning/lightning-bolts/blob/master/README.md

For which parts of the general README would you use this?

Haydnspass, 2021-03-22:
Maybe for the installation instructions? But I leave you the freedom, you have more of an outside view on this

Haydnspass, 2021-03-22:
I think the menu bar thing is solved like so: https://stackoverflow.com/a/40579324/10900884 or not?

phoess, 2021-03-22:
No, that unfortunately doesn't work. It generates a superlong file that is not integrated in the sidebar and the MD files still don't have a menu bar. I think the problem is mixing the two filetypes...

Haydnspass, 2021-03-22:
If you want you could convert the MD to rst
https://cloudconvert.com/md-to-rst

phoess, 2021-03-23:
I just pushed the converted files, I also changed the hierarchies a bit to get a nice TOC on the index site. Please check it out and let me know if you like it.

Haydnspass, 2021-03-23:
Looks nice. The only thing I am missing is the instructions to the instructions for SMAP? Or was that not possible?

phoess, 2021-03-23:
You mean a link to the installation instructions of SMAP? So far there is a link to rieslab.de in the Tutorial and it is mentioned that the docs and installation instructions can be found there. If you want I can put it more prominently, e.g. another subsection on the Installation page.

Haydnspass, 2021-03-23:
Yeah I meant the latter, the instructions for the bead calibration.

phoess, 2021-03-23:
Ok, I now linked both the User Guide and the Step by Step guide in my latest commit

closed 2021-03-23 by Haydnspass