TuragaLab/DECODE #106: Instructions for Testers and Collection of Feedback / Bugs for DECODE v0.10.rc.2

Haydnspass, 2021-03-24:
Dear Testers,

Thanks for taking the time to test DECODE. Here are the very short instructions to test decode. For all of your tests the following conditions hold:
- The basis of information is the README.md and the documentation https://decode.readthedocs.io/en/master/ (IMPORTANT: the master version, not the default page you get when you type it in without /master)
- If something is not self-explanatory please add it to this issue (or collect it elsewhere)
- If something does not work at all, contact me or someone of the DECODE team directly

Test Instructions:
- Is the readme instructive and friendly?
- Test the v0.10rc Colab Notebooks that you can find in the readme. Please add comments there directly. Note that it is okay that all of the cells are visible and not hidden (this will change upon release)
- Consult the docs (link above, again please use master) to install decode on your local machine
- Follow the tutorial

Feel free to add and collect comments here.

jries, 2021-03-25:
Colab:
Train:
1. Install and import DECODE: takes quite a while. Progress bar? Or warn (as for the download of example files)?
2. Tensor board not shown: it stays white. But it appears a while after starting the training. State this, otherwise people run this line many times.
3. I had problems running the training in Safari after I stopped it once. Then even reloading the page or restarting the run-time did not help. I then switched to chrome, but didn't do the restart, so not sure if also a problem there.
4. When stopping and restarting the training, the progress is displayed line by line (new line created for every update).
5. Some times the loss increases, the metrics produce NaNs. Is there a way to automatically restart training in this case or warn the user?
Fit:
6. Check random frame: the fluorophores are hardly visible, they are dominated by x for positions.

jries, 2021-03-25:
Workflow:
This starts with SMAP, bead calibration and training parameters. Please add above the link to the example files (for those who don't want to use own data). 
You can also state that this step is optional for the example files, and people can directly go to Training and fit.

jries, 2021-03-25:
Tutorial: training and fit.
Maybe list the notebooks and what they do (to give the user an idea which to start with, which to load)

cleterrier, 2021-03-25:
Trying the local install. Used the new command above to create a decode10_env environment and then went on to install using the master installation tutorial but I have the feeling that it's downgrading to pytorch 1.6.0, which is not compatible with my graphics card (RTX 3090)?

``(decode10_env) C:\Users\chris>conda install -c turagalab -c pytorch -c conda-forge decode
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: done
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: done

## Package Plan ##

  environment location: C:\Users\chris\.conda\envs\decode10_env

  added / updated specs:
    - decode

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    cudatoolkit-10.2.89        |       hb195166_8       398.2 MB  conda-forge
    pytorch-1.6.0              |py3.8_cuda102_cudnn7_0       704.7 MB  pytorch
    torchvision-0.7.0          |       py38_cu102         6.3 MB  pytorch
    ------------------------------------------------------------
                                           Total:        1.08 GB

The following NEW packages will be INSTALLED:

  torchvision        pytorch/win-64::torchvision-0.7.0-py38_cu102

The following packages will be SUPERSEDED by a higher-priority channel:

  decode             turagalab/label/rc::decode-0.10.0.rc.~ --> turagalab::decode-0.9.4-np118py_0
  spline             turagalab/label/rc::spline-0.10.0.rc.~ --> turagalab::spline-0.9.1-np118py38ha925a31_3

The following packages will be DOWNGRADED:

  cudatoolkit                             11.0.3-h3f58a73_8 --> 10.2.89-hb195166_8
  numpy                               1.19.5-py38h0cc643e_1 --> 1.18.5-py38h72c728b_0
  pytorch                      1.7.1-py3.8_cuda110_cudnn8_0 --> 1.6.0-py3.8_cuda102_cudnn7_0

Proceed ([y]/n)?``

Haydnspass, 2021-03-25:
@cleterrier What a nice surprise you are also testing :)
I figure you skipped my note above about deviating from the docs (only for the release candidate).
The command must be (notice the `-c turagalab/label/rc`)
```
conda create -n decode_env -c turagalab/label/rc -c pytorch -c conda-forge decode=0.10.0.rc.2 cudatoolkit jupyterlab ipykernel
```

Haydnspass, 2021-03-25:
bug reported by Sheng
```
Error using readtable (line 216)
Invalid parameter name: NumHeaderLines.
 
Error using readtable (line 216)
Invalid parameter name: NumHeaderLines.

Error in File.Load.Loader_decode>loadcsv (line 125)
locs=readtable(file,'NumHeaderLines',3);

Error in File.Load.Loader_decode/load (line 18)
                    [locs,info]=loadcsv(file);

Error in gui.GuiFile>loadfiles (line 332)
loader.load(par,[pfad f]);

Error in gui.GuiFile/loadbutton_callback (line 145)
                    loadfiles(obj,loader,f{k},pfad)

Error in gui.GuiFile>@(varargin)obj.loadbutton_callback(varargin{:}) (line 407)
pard.load.object=struct('Style','pushbutton','String','Load','FontWeight','bold','Callback',{{@obj.loadbutton_callback,0}});
 
Error while evaluating UIControl Callback.
```

cleterrier, 2021-03-25:
> @cleterrier What a nice surprise you are also testing :)
> I figure you skipped my note above about deviating from the docs (only for the release candidate).
> The command must be (notice the `-c turagalab/label/rc`)
> 
> ```
> conda create -n decode_env -c turagalab/label/rc -c pytorch -c conda-forge decode=0.10.0.rc.2 cudatoolkit jupyterlab ipykernel
> ```
> 
> Maybe you need to pin the cudatoolkit for the RTX 3090 as well --> `cudatoolkit=11.1`

I did create the environment with your new line, pinning the rc but I copy/pasted the following line (install decode) that I took from the master tutorial and its output

cleterrier, 2021-03-25:
Actually the environment creation command I used refered to rc.2 not just rc. Will strat from scratch in yet another environment.

Haydnspass, 2021-03-25:
@cleterrier I don't quite get it. After your first command 
```
conda create -n decode10_env -c turagalab/label/rc -c pytorch -c conda-forge decode=0.10.0.rc.2 cudatoolkit jupyterlab ipykernel
```
everything is there and seems fine. Why do you after activating the env do 
`conda install -c turagalab -c pytorch -c conda-forge decode`?

**If** you get the cuda version error because of your fancy gpu, maybe do 
`conda create -n decode10_env -c turagalab/label/rc -c pytorch -c conda-forge decode=0.10.0.rc.2 cudatoolkit=11.1 jupyterlab ipykernel`

cleterrier, 2021-03-25:
right, I shouldn't have done the last line!

cleterrier, 2021-03-25:
Second try - install works OK but i have to pin cudatoolkit 11.0, because 11.1 results in

```
(base) C:\Users\chris>conda create -n decode10_env -c turagalab/label/rc -c pytorch -c conda-forge decode=0.10.0.rc.2 cudatoolkit=11.1 jupyterlab ipykernel
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: |
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
Examining conflict for jupyterlab ipykernel cudatoolkit decode:  29%|█████▋              | 2/7 [00:00<00:02,  2.40it/s]/failed

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package python_abi conflicts for:
ipykernel -> pyzmq!=21 -> python_abi=2.7[build=*_cp27m]
decode=0.10.0.rc.2 -> h5py -> python_abi[version='2.7.*|3.6.*|3.8.*|3.9.*|3.7.*',build='*_cp27m|*_cp38|*_cp39|*_cp36m|*_cp37m']
ipykernel -> python_abi[version='3.6.*|3.7.*|3.8.*|3.9.*',build='*_cp36m|*_cp38|*_cp37m|*_cp39']
jupyterlab -> ipython -> python_abi[version='2.7.*|3.6.*|3.7.*|3.8.*|3.9.*',build='*_cp27m|*_cp36m|*_cp38|*_cp37m|*_cp39']

Package tornado conflicts for:
jupyterlab -> tornado[version='!=6.0.0,!=6.0.1,!=6.0.2|>=6.1|>=6.1.0']
jupyterlab -> notebook[version='>=4.3.1'] -> tornado[version='>=4|>=4,<6|>=4.1,<7|>=5.0|>=5.0,<7']

Package cudatoolkit conflicts for:
cudatoolkit=11.1
decode=0.10.0.rc.2 -> pytorch=1.7.1 -> cudatoolkit[version='>=10.1,<10.2|>=11.0,<11.1|>=10.2,<10.3']

Package python conflicts for:
jupyterlab -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.5,<3.6.0a0|>=3.5|>=3.6|>=3.7,<3.8.0a0|>=3.6,<3.7.0a0|3.4.*']
jupyterlab -> ipython -> python[version='>=2.7|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0|>=3']

Package python-dateutil conflicts for:
decode=0.10.0.rc.2 -> matplotlib -> python-dateutil[version='>=2.5.*|>=2.6.1|>=2.7.3']
ipykernel -> jupyter_client -> python-dateutil[version='>=2.1']

Package dataclasses conflicts for:
jupyterlab -> jupyter_server[version='>=1.4,<2'] -> dataclasses
decode=0.10.0.rc.2 -> pytorch=1.7.1 -> dataclasses

Package vc conflicts for:
cudatoolkit=11.1 -> vc[version='>=14.1,<15.0a0']
ipykernel -> python[version='>=3.7,<3.8.0a0'] -> vc[version='10.*|14.*|>=14.1,<15.0a0|>=14,<15.0a0|9.*|>=9,<10.0a0']
jupyterlab -> python[version='>=3.6'] -> vc[version='10.*|14.*|>=14.1,<15.0a0|>=14,<15.0a0|9.*|>=9,<10.0a0']
decode=0.10.0.rc.2 -> h5py -> vc[version='14.*|9.*|>=14,<15.0a0|>=14.1,<15.0a0']
Note that strict channel priority may have removed packages required for satisfiability.
```

cleterrier, 2021-03-25:
then, installing notebooks I get this:

```
(decode10_env) C:\Users\chris>python -m decode.utils.notebooks C:\Users\chris\christo\Processing\DECODE10
Traceback (most recent call last):
  File "C:\Users\chris\.conda\envs\decode10_env\lib\site-packages\git\__init__.py", line 83, in <module>
    refresh()
  File "C:\Users\chris\.conda\envs\decode10_env\lib\site-packages\git\__init__.py", line 73, in refresh
    if not Git.refresh(path=path):
  File "C:\Users\chris\.conda\envs\decode10_env\lib\site-packages\git\cmd.py", line 278, in refresh
    raise ImportError(err)
ImportError: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\chris\.conda\envs\decode10_env\lib\runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "C:\Users\chris\.conda\envs\decode10_env\lib\runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "C:\Users\chris\.conda\envs\decode10_env\lib\site-packages\decode\__init__.py", line 16, in <module>
    import decode.neuralfitter
  File "C:\Users\chris\.conda\envs\decode10_env\lib\site-packages\decode\neuralfitter\__init__.py", line 13, in <module>
    import decode.neuralfitter.train
  File "C:\Users\chris\.conda\envs\decode10_env\lib\site-packages\decode\neuralfitter\train\__init__.py", line 1, in <module>
    import decode.neuralfitter.train.random_simulation
  File "C:\Users\chris\.conda\envs\decode10_env\lib\site-packages\decode\neuralfitter\train\random_simulation.py", line 2, in <module>
    import decode.utils
  File "C:\Users\chris\.conda\envs\decode10_env\lib\site-packages\decode\utils\__init__.py", line 1, in <module>
    from . import bookkeeping
  File "C:\Users\chris\.conda\envs\decode10_env\lib\site-packages\decode\utils\bookkeeping.py", line 4, in <module>
    import git
  File "C:\Users\chris\.conda\envs\decode10_env\lib\site-packages\git\__init__.py", line 85, in <module>
    raise ImportError('Failed to initialize: {0}'.format(exc)) from exc
ImportError: Failed to initialize: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet

(decode10_env) C:\Users\chris>
```

Haydnspass, 2021-03-25:
Ah okay, I was not sure about the version, but then the specification without the exact version worked, right? :)

Haydnspass, 2021-03-25:
from Aline: Unfortunate default device in notebooks (used 'cuda:1', should be 'cuda:0')

phoess, 2021-03-26:
I'm getting the following error when fitting locally on Linux using the Jupyter Notebook

`  0%|          | 0/63 [00:00<?, ?it/s]

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-8-508cd961f27a> in <module>
      3                                   device=device, num_workers=worker)
      4 
----> 5 emitter = infer.forward(frames[:])

~/xconda3/envs/decode_env/lib/python3.8/site-packages/decode/neuralfitter/inference/inference.py in forward(self, frames)
     91                 so that we can easily concatenate."""
     92                 if self.post_proc is not None:
---> 93                     out.append(self.post_proc.forward(y_out))
     94                 else:
     95                     out.append(y_out.detach().cpu())

~/xconda3/envs/decode_env/lib/python3.8/site-packages/decode/neuralfitter/utils/processing.py in forward(self, *x)
     70                         x = com.forward(com_in)
     71                 else:
---> 72                     x = com.forward(*x)
     73             else:
     74                 x = com.forward(x)

~/xconda3/envs/decode_env/lib/python3.8/site-packages/decode/neuralfitter/scale_transform.py in forward(self, x)
    315         x = x.clone()
    316 
--> 317         x[:, 1] *= self.phot_max
    318         x[:, 5] *= self.phot_max  # sigma rescaling
    319 

TypeError: unsupported operand type(s) for *=: 'Tensor' and 'NoneType'`

phoess, 2021-03-26:
Concerning the last error: I used the wrong YAML file for `param_path` (the one generated for training the model), maybe it's worth mentioning in more detail which files are loaded in this notebook cell.

Haydnspass, 2021-03-27:
@phoess  I've added that to the fitting notebook

cleterrier, 2021-03-28:
I re-installed from scratch and it works with no special pinning of cudatoolkit for the RT3090.

Training notebook: there's a strange cropping of the real frames happening - I use an already cropped region of 40x40 pixels from an acquisition as real frames for test, and due to the change in the real frames plotting command , it ends up with just a small rectangular region plotted:

[image]

Haydnspass, 2021-03-28:
@cleterrier That's just due to the size of your frame. If you remove `30:70, -40:`
from the last imshow line you showed here you will get the complete frame.

closed 2021-03-28 by Haydnspass

cleterrier, 2021-03-28:
> @cleterrier That's just due to the size of your frame. If you remove `30:70, -40:`
> from the last imshow line you showed here you will get the complete frame.

I know it's because of this line, that's why I replaced it by the line below it (commented in the screenshot) without the `30:70, -40:`. But my point was shouldn't this `30:70, -40:` as default in the notebook be removed or modified if it leads to improper display for certain frames?