TuragaLab/DECODE #108: Notebook changes

Haydnspass, 2021-03-25:
- [ ] from Aline: Unfortunate default device in notebooks (used 'cuda:1', should be 'cuda:0')
- [ ] maybe CUDA check via cuda.is_available()?

Haydnspass, 2021-03-26:
- [ ] Add a comment if someone has already downloaded the example package because they wanted to follow from step 1 in smap

phoess, 2021-03-26:
- [ ] Note that example data has to be skipped if you want to use your own data in training notebook
- [ ] Similarly for YAML file to be loaded
- [ ] I had to enter the lab's standard password to access the notebooks - is this intended? I think we should change it.
- [ ] Typo: `Be aware that camera.forward() and camera.forward() …`: one of them should be `camera.backward()`
- [ ] Fit.ipynb: Why does filtering leave us with 74% of the detections? Isn’t this specific to the example file? => Mention! But curiously, even with my own data it’s 74% - Why?

Haydnspass, 2021-03-27:
The password is just because of your local installtion of jupyterlab, not related to the notebook.

Haydnspass, 2021-03-27:
@phoess you mean in the downlaoding stuff in the training notebook?

phoess, 2021-03-27:
Yes, but I don't find it very easy to follow the Training notebook with your own data. I'm wondering if it makes sense to have 2 notebooks there, one with own data and one with example data. But that would probably have too many duplications. Maybe having two cells at the important steps, one clearly for own data and one for example data helps with that?

Haydnspass, 2021-03-28:
> [ ] Fit.ipynb: Why does filtering leave us with 74% of the detections? Isn’t this specific to the example file? => Mention! But curiously, even with my own data it’s 74% - Why?

Thats just a statistical result because we filter 40nm and 80nm.

Haydnspass, 2021-03-29:
@phoess Is it better now?

phoess, 2021-03-29:
Yes, my points are all solved then.

closed 2021-03-29 by Haydnspass