TuragaLab/DECODE #117: Errors while saving emitter as csv

cleterrier, 2021-03-31:
Hi,

I have a notebook to batch filter emitters that I have updated for 0.10. It takes h5 as input (used to take pt with 0.9.4)and saves filtered emitters as csv. Here is the relevant code:

`for h5_path in filelist : 

    h5_path = h5_path.replace("\\", "/")
    print("processing file " + h5_path)

    #Load the .h5 file
    emitter = decode.EmitterSet.load(h5_path) # path to the h5 file
    
    csvout_path = h5_path.replace(".h5", ".csv") 
    print('filtered csv will be saved as: ' + csvout_path)
    
    # filter on uncertainties (absolute) for EMCCD
    sigma_x_high_threshold = 25 # be aware that these values will be *0.4 when converting into TS, like 50 = 20 nm, 22.5 = 9 nm
    sigma_y_high_threshold = 25 
    sigma_z_high_threshold = 100
    
    em_sub = emitter[(emitter.xyz_sig_nm[:, 0] <= sigma_x_high_threshold) * (emitter.xyz_sig_nm[:, 1] <= sigma_y_high_threshold) * (emitter.xyz_sig_nm[:, 2] <= sigma_z_high_threshold)]
    
    # Save filtered emitters as csv
    csvout_path = h5_path.replace(".h5", ".csv") 
    #decode.utils.emitter_io.save_csv(csvout_path, em_sub.to_dict())
    em_sub.save('csvout_path')
    print('saved filtered csv as: ' + csvout_path)`

Haydnspass, 2021-03-31:
Can you show me the error? 

One thing: ```#decode.utils.emitter_io.save_csv(csvout_path, em_sub.to_dict())``` will not work anymore because the signature has changed https://decode.readthedocs.io/en/release-0.10/decode.utils.html#decode.utils.emitter_io.save_csv
But you are calling .save directly, so that is unrelated

cleterrier, 2021-03-31:
Sorry I was completing/editing the post. Should be complete with the two errors now

Haydnspass, 2021-03-31:
You are parsing a string to `.save()` it should be `em_sub.save(csvout_path)` (i.e. without marks).

You are still free to use the emitter_io.save_csv function, but you would need to call it like this
```python
decode.utils.emitter_io.save_csv(path, em_sub.data, em_sub.meta):
```

cleterrier, 2021-03-31:
Ha! Sorry for that

closed 2021-03-31 by Haydnspass

Haydnspass, 2021-03-31:
No worries :P