TuragaLab/DECODE #130: ImportError when trying to run Step 1: Install and Import DECODE

bellsrockets, 2021-07-08:
I am trying to run Step 1 of the setup of the Training notebook, and get the following error message:

ImportError: cannot import name 'container_abcs' from 'torch._six' (/usr/local/lib/python3.7/dist-packages/torch/_six.py)

I'm guessing this is due to changes in the torch libraries with the newest version of PyTorch (v1.9.0), but I'm not sure how best to fix it. 

Many thanks for your help!

Haydnspass, 2021-07-08:
Thanks for pointing this out!
Please fix the pytorch version during installation I.e.
"Conda install -c [...] decode pytorch=1.7.1"

PierreMangeol, 2021-07-08:
Would it be possible to fix this issue on the Colab tutorials?

Thanks for your help!

Haydnspass, 2021-07-09:
Yes of course. We will be on it soon!

Haydnspass, 2021-07-10:
@PierreMangeol Fixed on Colab. Locally it should not have appeared. If you experienced otherwise please comment or open a new issue :)

closed 2021-07-10 by Haydnspass

PierreMangeol, 2021-07-10:
Perfect, thanks!