TuragaLab/DECODE #168: Trouble Installing torch 1.7.1 on Colab

frankfazekas, 2022-04-02:
When running the first cell of the "Training a DECODE model" on Colab, I receive the error `ValueError: module functions cannot set METH_CLASS or METH_STATIC` in line 190 of `/usr/local/lib/python3.7/dist-packages/torch/__init__.py`.  It seems to be an issue with installing torch. I tried installing 1.8.1 instead but received a similar error.  Is there another version I should be trying?

Haydnspass, 2022-04-04:
Hey @frankfazekas thanks for reporting this, I will look into it and get back to you.
Best,
Lucas

closed 2022-04-16 by frankfazekas

Haydnspass, 2022-04-17:
Dear @frankfazekas I belive this has been resolved? I have tried the notebooks and could not reproduce the bug?

frankfazekas, 2022-04-17:
Dear Lucas, yes, thanks for looking into it. I'm not sure what the underlying issue was, but I haven't experienced the problem since. I am currently training a model to apply to some of my own data.