TuragaLab/DECODE #177: TypeError: imread() got unexpected keyword arguments 'multifile'

HypnosWei, 2022-09-23:
When I use the Training.ipynb, this part seems to go something wrong 

'''
tar_em, sim_frames, bg_frames = simulator.sample()
sim_frames = sim_frames.cpu()

frame_path = zip_folder   # change if you load your own data
data_frames = decode.utils.frames_io.load_tif(frame_path).cpu()

print(f'Data shapes, simulation: {sim_frames.shape}, real data: {data_frames.shape}')
print(f'Average value, simulation: {sim_frames.mean().round()}, real data: {data_frames.mean().round()}')
'''

when I run this cell, it appeals to this TypeError: imread() got unexpected keyword arguments 'multifile'
so I find the document of tifffile, I find it said 'Remove deprecated multifile parameters from TiffFile (breaking).'

JohannaRahm, 2022-09-26:
I get the same error, which seems to be related to the tifffile version. The error is caused with version 2022.8.12, but no error appears with version 2022.4.8.

A quick fix is to create an environment with set tifffile version
`conda create -n decode_env2 -c turagalab -c pytorch -c conda-forge decode=0.10.0 cudatoolkit=11.0 tifffile=2022.4.8 jupyterlab ipykernel`

Perhaps the package version could be restricted in the environment.yaml file?

HypnosWei, 2022-09-27:
> I get the same error, which seems to be related to the tifffile version. The error is caused with version 2022.8.12, but no error appears with version 2022.4.8.
>
> A quick fix is to create an [...]

Thank you! it works, by the way, did you ever try the 2D images super-resolution on decode?

Haydnspass, 2022-10-04:
@HypnosWei @JohannaRahm Sorry for the slow response, I overlooked this issue.
Yes, indeed the tifffile version can be fixed. 
However there is some trouble with this package, but your command works. I will take a look and possibly pin it in the release.

Haydnspass, 2023-02-21:
fixed by release 0.10.1

closed 2023-02-21 by Haydnspass