TuragaLab/DECODE #178: scipy version 1.9.1 error while training

JohannaRahm, 2022-09-28:
Hi,

training throws a ValueError within the scipy library (version 1.9.1). The process runs through with scipy version 1.8.0. There might be breaking changes in the newer scipy library.

**Error message:**

[image]

HypnosWei, 2022-09-29:
I get the same error, I try to uninstall scipy 1.9.0, and install scipy 1.8.0, but the decode package just disappears...

JohannaRahm, 2022-09-29:
a workaround is to create a new environment 

`conda create -n decode_env2 -c turagalab -c pytorch -c conda-forge decode=0.10.0 cudatoolkit=11.0 tifffile=2022.4.8 scipy=1.8.0 jupyterlab ipykernel`

the tifffile version is fixed because of this issue #177

Haydnspass, 2022-10-04:
Thanks for pointing this out, I will pin it in a patch release.

Haydnspass, 2022-10-25:
@HypnosWei @JohannaRahm if you want you can install the patched version as

```bash
conda create -n decode_v0_10_1rc1 -c turagalab/label/rc -c turagalab/label/dev -c pytorch -c conda-forge decode=0.10.1 cudatoolkit=11.3 jupyterlab ipykernel
```

for me everything works now, happy to get your feedback :)

JohannaRahm, 2022-10-25:
this version runs through without errors (intro, train, fit, evaluation notebooks and training command via CLI was tested). A minor thing: Compared to the earlier version there are a lot of warnings appearing while training through CLI. The warnings consume quite a lot of output space and the progress bar is a bit lost in it. I'll post the warnings here in case you want to take care of them for future updates.

Haydnspass, 2023-02-21:
It was fixed by release 0.10.1 from yesterday.

closed 2023-02-21 by Haydnspass