TuragaLab/DECODE #216: Fitting.ipynb to fitting.py

Ryan-Gh1, 2023-05-18:
How does the fitting file work in pycharm?
I transform the fitting.ipynb to fitting.py,but it can't run in the pychram IDE.
It throws a mistake in line 'emitter = infer.forward(frames[:])'.
It says that "TypeError: conv2d() received an invalid combination of arguments - got (Tensor, Parameter, Parameter, tuple, tuple, tuple, int), but expected one of:
 * (Tensor input, Tensor weight, Tensor bias, tuple of ints stride, tuple of ints padding, tuple of ints dilation, int groups)
      didn't match because some of the arguments have invalid types: (Tensor, !Parameter!, !Parameter!, !tuple!, !tuple!, !tuple!, int)"
      It's important for me,so please deal with it soon.Thanks a lot from my heart for your effort!

Haydnspass, 2023-06-19:
@Ryan-Gh1 sorry I missed your message.
Could you provide more information how you installed DECODE, which version etc.?

Ryan-Gh1, 2023-06-20:
> @Ryan-Gh1 sorry I missed this issue. Could you provide more information how you installed DECODE, which version etc.?

The version is 0.10.2.  I follow the tutorial.
I solved it by reinstalling the environment,I don't know why it appears.

By the way,I have some problems about the corresponding paper.
Would you please tell me how to calculate the emitters' density?
Thanks a lot again.

Haydnspass, 2023-06-22:
> > @Ryan-Gh1 sorry I missed this issue. Could you provide more information how you installed DECODE, which version etc.?
> 
> The version is 0.10.2. I follow the tutorial. I solved it by reinstalling the environment,I don't know why it appears.
okay good!

> By the way,I have some problems about the corresponding paper. Would you please tell me how to calculate the emitters' density? Thanks a lot again.

could you open a discussion for that? :) closing this issue because these are two different things

closed 2023-06-22 by Haydnspass