TuragaLab/DECODE #217: Install numba in DECODE environment?

cleterrier, 2023-06-09:
Hi,
I am trying to locally install the COMET drift-correction notebook (https://github.com/gpufit/Comet/blob/master/Colab_notebooks/COMET.ipynb) to chain it with DECODE output and correct drift on localization data.

The "from numba import cuda" fails because numba is not intalled in the DECODE environment. I tried a simple "conda install numba" but it does not seem to resolve properly. Any idea of what I should do?

Here is everything that is installed in my current DECODE envrionment:
`
absl-py                   0.12.0             pyhd8ed1ab_0    conda-forge
aiohttp                   3.7.4            py38h294d835_0    conda-forge
anyio                     2.2.0            py38haa244fe_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
argon2-cffi               20.1.0           py38h294d835_2    conda-forge
async-timeout             3.0.1                   py_1000    conda-forge
async_generator           1.10                       py_0    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     20.3.0             pyhd3deb0d_0    conda-forge
babel                     2.9.0              pyhd3deb0d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
blas                      2.106                       mkl    conda-forge
bleach                    3.3.0              pyh44b312d_0    conda-forge
blinker                   1.4                        py_1    conda-forge
blosc                     1.21.0               h0e60522_0    conda-forge
brotli                    1.0.9                h0e60522_4    conda-forge
brotlipy                  0.7.0           py38h294d835_1001    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2020.12.5            h5b45459_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cachetools                4.2.1              pyhd8ed1ab_0    conda-forge
certifi                   2020.12.5        py38haa244fe_1    conda-forge
cffi                      1.14.5           py38hd8c33c5_0    conda-forge
chardet                   4.0.0            py38haa244fe_1    conda-forge
charls                    2.2.0                h39d44d4_0    conda-forge
click                     7.1.2              pyh9f0ad1d_0    conda-forge
cloudpickle               1.6.0                      py_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
cryptography              3.4.7            py38hd7da0ea_0    conda-forge
cudatoolkit               11.0.3               h3f58a73_8    conda-forge
cycler                    0.10.0                     py_2    conda-forge
cytoolz                   0.11.0           py38h294d835_3    conda-forge
dask-core                 2021.4.1           pyhd8ed1ab_0    conda-forge
decode                    0.10.0                np119py_0    turagalab
decorator                 4.4.2                      py_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
deprecated                1.2.12             pyh44b312d_0    conda-forge
deprecation               2.1.0              pyh9f0ad1d_0    conda-forge
entrypoints               0.3             pyhd8ed1ab_1003    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
fsspec                    2021.4.0           pyhd8ed1ab_0    conda-forge
giflib                    5.2.1                h8d14728_2    conda-forge
gitdb                     4.0.7              pyhd8ed1ab_0    conda-forge
gitpython                 3.1.15             pyhd8ed1ab_0    conda-forge
google-auth               1.28.0             pyh44b312d_0    conda-forge
google-auth-oauthlib      0.4.1                      py_2    conda-forge
grpcio                    1.37.0           py38he5377a8_0    conda-forge
h5py                      3.2.1           nompi_py38he6c2248_100    conda-forge
hdf5                      1.10.6          nompi_h5268f04_1114    conda-forge
icu                       68.1                 h0e60522_0    conda-forge
idna                      2.10               pyh9f0ad1d_0    conda-forge
imagecodecs               2021.3.31        py38hccd6b70_0    conda-forge
imageio                   2.9.0                      py_0    conda-forge
importlib-metadata        4.0.1            py38haa244fe_0    conda-forge
importlib_resources       5.1.2            py38haa244fe_0    conda-forge
iniconfig                 1.1.1              pyh9f0ad1d_0    conda-forge
intel-openmp              2021.2.0           h57928b3_616    conda-forge
ipykernel                 5.5.3            py38h43734a8_0    conda-forge
ipython                   7.22.0           py38h43734a8_0    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
[... 139 lines cut ...]
traitlets                 5.0.5                      py_0    conda-forge
typing-extensions         3.7.4.3                       0    conda-forge
typing_extensions         3.7.4.3                    py_0    conda-forge
urllib3                   1.26.4             pyhd8ed1ab_0    conda-forge
vc                        14.2                 hb210afc_4    conda-forge
vs2015_runtime            14.28.29325          h5e1d092_4    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
webencodings              0.5.1                      py_1    conda-forge
werkzeug                  1.0.1              pyh9f0ad1d_0    conda-forge
wheel                     0.36.2             pyhd3deb0d_0    conda-forge
widgetsnbextension        3.5.1                    pypi_0    pypi
win_inet_pton             1.1.0            py38haa244fe_2    conda-forge
wincertstore              0.2             py38haa244fe_1006    conda-forge
winpty                    0.4.3                         4    conda-forge
wrapt                     1.12.1           py38h294d835_3    conda-forge
xmltodict                 0.12.0                   pypi_0    pypi
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0    conda-forge
yarl                      1.6.3            py38h294d835_1    conda-forge
zeromq                    4.3.4                h0e60522_0    conda-forge
zfp                       0.5.5                h0e60522_5    conda-forge
zipp                      3.4.1              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.4.9                h6255e5f_0    conda-forge
`

cleterrier, 2023-06-09:
Here's the end of the output of "conda install numba":

The following specifications were found to be incompatible with your system:

  - feature:/win-64::__win==0=0
  - feature:|@/win-64::__win==0=0
  - conda-forge/noarch::google-auth-oauthlib==0.4.1=py_2 -> click -> __unix
  - conda-forge/noarch::google-auth-oauthlib==0.4.1=py_2 -> click -> __win
  - conda-forge/noarch::jupyterlab==3.0.14=pyhd8ed1ab_0 -> ipython -> __linux
  - conda-forge/noarch::jupyterlab==3.0.14=pyhd8ed1ab_0 -> ipython -> __win
  - conda-forge/noarch::notebook==6.3.0=pyha770c72_1 -> ipykernel -> __linux
  - conda-forge/noarch::notebook==6.3.0=pyha770c72_1 -> ipykernel -> __win
  - conda-forge/noarch::urllib3==1.26.4=pyhd8ed1ab_0 -> pysocks[version='>=1.5.6,<2.0,!=1.5.7'] -> __unix
  - conda-forge/noarch::urllib3==1.26.4=pyhd8ed1ab_0 -> pysocks[version='>=1.5.6,<2.0,!=1.5.7'] -> __win
  - conda-forge/win-64::ipykernel==5.5.3=py38h43734a8_0 -> ipython[version='>=5.0'] -> __linux
  - conda-forge/win-64::ipykernel==5.5.3=py38h43734a8_0 -> ipython[version='>=5.0'] -> __osx
  - conda-forge/win-64::ipykernel==5.5.3=py38h43734a8_0 -> ipython[version='>=5.0'] -> __win
  - conda-forge/win-64::jupyter_server==1.6.4=py38haa244fe_0 -> send2trash -> __linux
  - conda-forge/win-64::jupyter_server==1.6.4=py38haa244fe_0 -> send2trash -> __win
  - conda-forge/win-64::pysocks==1.7.1=py38haa244fe_3 -> win_inet_pton -> __win
  - conda-forge/win-64::setuptools==49.6.0=py38haa244fe_3 -> wincertstore[version='>=0.2'] -> __win
  - turagalab/noarch::decode==0.10.0=np119py_0 -> click -> __unix
  - turagalab/noarch::decode==0.10.0=np119py_0 -> click -> __win

Your installed version is: 0

Note that strict channel priority may have removed packages required for satisfiability.

Haydnspass, 2023-06-19:
@cleterrier hmm, that's a good point. I don't think we can do anything about it in the released DECODE version, as conda states that the packages are, in fact, incompatible; probably because of some side requirements.
As a workaround you could save your localizations and then load them in a new environment with for the drift correction. 
The .h5 files are fairly easy to read, you can just take a look at the DECODE .h5 reader.

In a future release I might be able to change the requirements and/or relax them, such that combining both DECODE and the drift correction is possible.

cleterrier, 2023-06-27:
@Haydnspass thanks for the tip, I'm trying to set up an independent environment for this. I'm wondering how I can get the relevant variables from an emitter set (x,y,z coordinates) into arrays that are the entry format of COMET? Relevant import code is currently this:

`import numpy as np 
import pandas as pd 
import io

filepath = 'C:/Users/chris/Desktop/test.csv'

data = pd.read_csv(filepath)

localizations = np.zeros((len(data['frame']), 4))
localizations[:, 0] = np.asarray(data['x [nm]'])
localizations[:, 1] = np.asarray(data['y [nm]'])
localizations[:, 2] = np.asarray(data['z [nm]'])
localizations[:, 3] = np.asarray(data['frame'])
frames = np.unique(localizations[:, -1])
n_frames = len(frames)
print(f"{filepath} import successful, {len(localizations[:, 0])} localizations, {n_frames} frames")`

So basically I'd need a way to feed x,y,z coordinates as arrays after opening the h5 file. Is there a command for that?

Haydnspass, 2023-07-11:
I will have to look into that

nolan1999, 2023-07-11:
@cleterrier This should put x, y, z, and frame index from a .h5 file into a Nx4 numpy array:
`
import h5py
filename = <path_to_emitters.h5>

with h5py.File(filename, "r") as f:
    x, y, z = np.array(f["data"]["xyz"]).T
    f = np.array(f["data"]["frame_ix"])
    localizations = np.stack([x, y, z, f]).T
`
Does this help you?

closed 2023-08-28 by Haydnspass